dracoventions/TWCManager #6: Timed out waiting for response from TWCManager script

chran28, 2019-04-10:
Hello, I'm still a freshman in programming. Have the TWCManager installed according to the instructions. When I go to the website, I get the following error: "If the script is running, make sure the $ twcScriptDir parameter in the source of this web page points to the directory containing the TWCManager script.". I checked the parameter and am of the opinion that everything is correct. What can I do to solve the problem?

flodorn, 2019-04-10:
do you see the script running when you type "screen -r" on the rpi?

chran28, 2019-04-11:
Hello,

I do not see a script when I enter "screen -r". I get the answer "There is no screen to be resumed." Did the installation as in your description again. I checked the file in the folder home / pi / TWC / index.php, is that correct? Does it make a difference if I set the country code to DE UTF -8?

flodorn, 2019-04-11:
This means the script isn't running on the rpi and you have to set it up properly first. It is not an issue of changing the country code. 
Try running the script manually by typing "python3 TWC/TWCManager.py". If this works, you probably didn't setup the autostart bit properly.

chran28, 2019-04-11:
ok, the program says that the TWC is set as the master. The rotary switch is at F.

 

pi@TWCManager:~ $ python3 TWC/TWCManager.py

TWC Manager starting as fake Master with id 7777 and sign 77

09:28:02: Send master linkready1

09:28:02 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

09:28:02: Send master linkready1

09:28:02 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

09:28:02: Send master linkready1

09:28:02 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

09:28:02: Send master linkready1

09:28:02 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

09:28:02: Send master linkready1

09:28:03 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

09:28:03: Send master linkready2

09:28:03: 32.00 amp slave TWC 02BB is ready to link.  Sign: 2D

09:28:03: Set slave TWC 02BB protocolVersion to 1, minAmpsTWCSupports to 5.

09:28:03: Send master linkready2

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:03: Send master linkready2

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:03: Send master linkready2

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:03: Send master linkready2

09:28:03: SHB 02BB: 02 00.00/00.00A 0000  M: 05 00.00/00.00A 0000

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:03: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:04: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

[... 166 lines cut ...]
Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:47: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:48: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:49: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:50: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:51: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 05 00 00 00 00 00 00 00 00 90

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

09:28:52: WARNING: We haven't heard from slave 02BB for over 26 seconds.  Stop sending them heartbeat messages.

09:29:03 ERROR: Can't determine current solar generation from:

flodorn, 2019-04-11:
It looks like your wall connecter is set to master, not slave. You should disconnect the power and check that the rotary switch and dip switch are set correctly, see installation manual: https://www.tesla.com/sites/default/files/pdfs/wall-connector-eu/20161208/tesla-32a-wall-connector-installation-manual-en-EU-v2.pdf?201612081439

chran28, 2019-04-11:
The connector is set to slave.

flodorn, 2019-04-11:
Could be an issue with the wall connector. I'd set debugLevel to 11 in TWCManager.py and try different rotary switch positions on the wall connector to see which messages you get. Don't forget to power off the wall connector before making changes on the rotary switch. 

Tx@ is for messages you send to your wall connector 
Rx@ is for messages you receive

You could also try to run TWCManager as slave to see if this works.

chran28, 2019-04-11:
I also set the debug level to 11. Enclosed the protocol

  

 

pi@TWCManager:~ $ python3 TWC/TWCManager.py

TWC Manager starting as fake Master with id 7777 and sign 77

10:48:38: Send master linkready1

Tx@10:48:38: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0

Rx@10:48:38: () FC E1 77 77 77 00 00 00 00 00 00 00 00 46

10:48:38 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

10:48:38: Send master linkready1

Tx@10:48:38: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0

Rx@10:48:38: () FC E1 77 77 77 00 00 00 00 00 00 00 00 46

10:48:38 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

10:48:38: Send master linkready1

Tx@10:48:38: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0

Rx@10:48:38: () FC E1 77 77 77 00 00 00 00 00 00 00 00 46

10:48:38 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

10:48:38: Send master linkready1

Tx@10:48:38: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0

Rx@10:48:38: () FC E1 77 77 77 00 00 00 00 00 00 00 00 46

10:48:38 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

10:48:38: Send master linkready1

Tx@10:48:38: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0

Rx@10:48:39: () FC E1 77 77 77 00 00 00 00 00 00 00 00 46

10:48:39 ERROR: TWC is set to Master mode so it can't be controlled by TWCManager.  Search installation instruction PDF for 'rotary switch' and set switch so its arrow points to F on the dial.

10:48:39: Send master linkready2

Tx@10:48:39: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0

Rx@10:48:39: () FB E2 77 77 77 00 00 00 00 00 00 00 00 47

10:48:39: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

10:48:39: Send master linkready2

Tx@10:48:39: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0

Rx@10:48:39: () FB E2 77 77 77 00 00 00 00 00 00 00 00 47

10:48:39: *** UNKNOWN MESSAGE FROM SLAVE:FB E2 77 77 77 00 00 00 00 00 00 00 00 47

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

[... 366 lines cut ...]
Tx@10:50:52: C0 FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B C0

Rx@10:50:52: () FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

10:50:52: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

Tx@10:50:53: C0 FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B C0

Rx@10:50:53: () FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

10:50:53: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

Tx@10:50:54: C0 FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B C0

Rx@10:50:54: () FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

10:50:54: *** UNKNOWN MESSAGE FROM SLAVE:FB E0 77 77 02 BB 00 00 00 00 00 00 00 00 00 8B

Please private message user CDragon at http://teslamotorsclub.com with a copy of this error.

10:50:55: WARNING: We haven't heard from slave 02BB for over 26 seconds.  Stop sending them heartbeat messages.

chran28, 2019-04-11:
Because of the autostart, a picture of the parameters. Is it correct that way?

flodorn, 2019-04-11:
I don't see a picture. 

Your serial port gets the excact same repsonse as the message you send during linkready1, this is strange. It casues the TWCManager script to think that the wall connector is sending it and is a master, but it's just a repeated signal on the serial port, or even the same signal as the source. Could be an issue with the serial adapter or the wiring. 

How did you connect the serial wires? 

I'd also try resetting the wall connector by pressing the reset button for >10sec

chran28, 2019-04-11:
I have connected from the interface D- (RX) to D- in TWC and D + (TX) to D + in TWC. This is my interface [4.988866] usb 1-1.1.2: FTDI USB Serial Device Converter now attached to ttyUSB0. Picture is attached to the email. When I execute the command "python3 TWC / TWCManager.py" to the PI, the script runs. The command screen -r but nothing appears to me.

flodorn, 2019-04-11:
There is no image file attached to the email. 

As mentioned before, I'd try different settings on the rotary switch, change the serial cable from IN to OUT on the wall connector and run TWCManager as slave.

As for the autostart, if you followed the instructions of CDragon regarding the rc.local file it should work. I personally also had to change chmod of the TWC folder to 777 to make autostart work. 

If the script has an error it also won't start. But you ruled that out by running it direclty with the python3 command.

neoof86, 2019-05-02:
Good Evening,

I am having a similar issue, I have been setting this up as a POC until my PW arrive and was going to see if I could write an API to interact with my SolarEdge inverter for now but upon reboot the script just does not seem to run

Below is an example of the start up script, I have checked for any potential errors by copying from the PDF and ensuring this went in as plain text like below

[image]

I also tried to modify the script to try and run the following su - pi -c "screen -dm -S TWCManager ~/TWC/TWCManager.py" and still no luck, I also tried to also see if it was a permissions issue by amending the folder using chmod 777 but again it still would not start 

I have tried the command manually and get the following

[image]

If I run the Python the script runs as expect 

[image]

Whilst this is running it works as expected, Is there a simple step I have missed?

[image]

flodorn, 2019-05-02:
@neoof86 
The only difference I notice is that I added the startup bit `su - pi -c "screen -dm -S TWCscript ~/TWC/TWCManager.py"` in rc.local directly above the exit 0 line

juamiso, 2020-04-23:
I have set up the script as systemd service and is working properly (starting at boot). To do so, I did follow the guideline in [Systemd](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/) (method4).

create the file TWCManager.service in /etc/systemd/system/
the content should be:
`[Unit]
Description=Autostart TWCManager, logs to be found in journalctl -U TWCManager.system
 
[Service]
User=pi
Type=idle
ExecStart=/usr/bin/python3 /home/pi/TWC/TWCManager.py
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target
`
then give the proper rights, enable, start and reboot:
`sudo chmod 644 /lib/systemd/system/TWCManager.service
sudo systemctl daemon-reload
sudo systemctl enable TWCManager.service
sudo reboot`

If you want to check the status of the process, just write

`sudo systemctl status TWCManager.service`
you can "stop" it "restart" it or many other stuff to debug things. 

The logs are to be found using journalctl 

Maybe worth a try ?

simotronic, 2020-04-24:
Thanks,
That is the proper way to have it work. Not sure why local.rc was used?
Maybe a Raspian thing. I think it would be possible to add a cron job that
ran every so often to check it's running, if not, then start it. I
currently have the as suggested method and it occasionally stops! I'm
hoping to get back on track with this project to see if I can have all 3
HPTWC fighting over 80A!! and include an OpenEVSE just to make sure I'm
insane!!
Simon

-- 
This email, and any attachment to it, may contain information that is
proprietary, privileged or confidential or that may be otherwise legally
exempt from disclosure and is intended only for the individual(s) or entity
to which it is addressed.   If you are not the named recipient, or the
employee or agent responsible for delivering it to the intended recipient,
you are not authorized to read, print, retain, copy, disclose or distribute
this email or any part of it.
If you have received this email in error, please return it immediately to
the sender, delete it and all copies from your system, and destroy any hard
copies of this communication.