dracoventions/TWCManager #8: Stop car from charging

Tschingis, 2019-06-17:
Thanks for your great work. This is not directly an issue but more something i found out. As you have written in the code, protocol 2 does not allow to stop a car from charging. Using the car api does not work for me since there are non tesla cars charging at the twc. By accident i have seen that if the twcmanager script was not running, the twc goes into a resumable error state (Green on - 4 Red flashes) and it does not charge the car. It even stops the charge of the car, if you terminate the script during a charge. I made a dirty hack into the script which does not send a master heartbeat to a slave twc if it is not allowed to charge. I cannot test if this hack works with several slaves, i.e. if a slave keeps charging if he receives a heartbeat that is not adressed to him, but in case of one slave (one twc) it reproducable prevents charging of the car. If the heartbeat starts again, the charging of the car resumes.

infernix, 2019-09-04:
@Tschingis would you mind forking this and committing your hack? Or at least paste a diff here?

laur3ns, 2019-09-24:
I believe the trouble with this approach is that if the time window in which it is not receiving heartbeats is too long, the TWC and/or car (the latter if I recall) will go into a permanent error state, requiring a re plug to clear.

db9carbon, 2021-01-29:
> Thanks for your great work. This is not directly an issue but more something i found out. As you have written in the code, protocol 2 does not allow to stop a car from charging. Using the car api [...]

I too am looking for this, as using TWC on non-Tesla's too. Can you pls post your code?