dracoventions/TWCManager #10: Stuck Charging

PCRover, 2019-09-03:
Yesterday while charging my M3 the Tesla API went offline and TWC could no longer communicate with my car. The car kept charging at the last rate and I couldn't stop it besides unplugging the charge cable. Even turning off charging on the cars display only caused charging to stop, then a few seconds later start back up. 

Anyone have an idea on how to stop the TWC from charging via TWCManager when loosing connection to the car? I tried setting scheduled power to 0 and unscheduled power to 0. This had no effect in operation despite the web interface showing 0 available amps, the car continued to charge.

simotronic, 2019-09-04:
Maybe when the car is connected it could communicate directly to the TWC
and set a lower current? But maybe the car needs to negotiate with the
master? I haven't looked closely at the Tesla to TWC piece. I'm thinking
that's the PLC (Power Line Communication)?
Simon

deece, 2019-11-24:
I'm seeing something similar after a tech came over/latest firmware update (they were close together in time, so I'm not sure what caused it). I'm successfully tracking my surplus solar, but the car continues to charge after there is no longer enough solar.