dracoventions/TWCManager #12: How can i change ''Max amps available'' ?

DennisM12, 2019-09-30:
Dear,

My setup is almost working but I can't get more than '8 amps available', how can I change this  ? When I select 13A for example nothing changes and the car keeps charging on just 6A (I can stop and start the charging process so I have communication) but I want to charge the car faster and its possible hardware wise. I have 'wiringMaxAmpsAllTWCs' at 40 so that shouldnt be an issue. Anyone knows how to solve this ?

Thanks in advance.

Best regards,

Dennis

closed 2019-09-30 by DennisM12

robot256, 2019-09-30:
I don't think it's an issue with this project, but I had the same thing happen to my Model 3.  After a few hours of having my home automation system adjust the TWC power and turning the car on and off, the car's internal amp limit got stuck at 6 no matter what the TWC did.  I had to replug the connector to reset it.  Manually changing the limit might have worked to.  Seems like it's a bug and/or feature in the car software.  I'm working on a solar power tracking algorithm that involves fewer commands sent to the car to see if that helps.

DennisM12, 2019-09-30:
> I don't think it's an issue with this project, but I had the same thing happen to my Model 3. After a few hours of having my home automation system adjust the TWC power and turning the car on and [...]

Allready fixed!

robot256, 2019-10-13:
> Allready fixed!

May I ask what you did to fix it, or how you determined it is or is not the fault of TWCManager?  I'm still having this issue with the latest code from the repository.

deece, 2019-12-03:
> > Allready fixed!
> 
> May I ask what you did to fix it, or how you determined it is or is not the fault of TWCManager? I'm still having this issue with the latest code from the repository.

Are you limiting the charge current in-car?