dracoventions/TWCManager #14: Conditional modes

MikeBishop, 2019-10-18:
I'm interested in using this project for a slightly different purpose.  I'm okay if the car charges off the grid, so long as there is grid power -- the joy of net metering.  What I want is for the limitation to kick in when the power is off, to ensure that the car is never siphoning from the Powerwall.  So:

- If grid is up, full power is permitted.
- If grid is down:
    - Do not charge / provide minimum allow current until Powerwall recharges to X%
    - Track green energy until power is restored

laur3ns, 2019-10-18:
Should be do-able. You will have to do a few things
- query the different sources for their 'state': grid on/off and consumption; solar generation; powerwall state
- change the TrackGreenEnergy function to implement your policy and provide a singular 'available Amps for charging'.

I did a (slightly less complex) thing to track my home consumption and give my chargers (2) the remainder of the available amps as to now blow my main fuse.

ngardiner, 2019-11-02:
@MikeBishop if you don't have much luck with this, you can feel free to drop an issue on my fork at https://github.com/ngardiner/TWCManager, I've done quite a bit of modularisation of the TWCManager code including a module for Powerwall, it wouldn't take much to track grid status from the Powerwall API and use it as a condition - however not having a Powerwall myself I'd need your help testing it out.

MikeBishop, 2019-11-04:
Thanks for pointing to your fork, @ngardiner!  I realized after posting the issue that the base repo doesn't have any Powerwall-specific code and had found the TMC post that contained some, but I hadn't found your fork.

After a cursory look at your new "charging policy" code, I think this fits pretty well.  The "track green energy" path should be fine when running off-grid -- consume nothing if the PW battery is below target charge, consume excess solar output once the PW is charged.  The simplest change would be to make the "scheduled charging" function always say "not scheduled" if the grid is down.  Perhaps in a generic sense, it would make sense for the EMS modules to have a "veto scheduled charging" function?  That would even let you have the option of using "non-scheduled charging" to control what power draw you think is sustainable when running on battery.

My install isn't likely to be for another few months; I'm just figuring out what pieces I'll need to have to automate it.  Happy to try things out with you once I have hardware!