dracoventions/TWCManager #16: No slave TWCs found on RS485 network.

llamad911, 2019-11-19:
I followed the instructions and when I open the browser to the IP adress, I have these messages:
Power available for all TWCs: None
No slave TWCs found on RS485 network.

The rotary selector of my Wall connector is set to "F" slave.
I have the top green led on and a 4 time flashing red led.
The user manual say: Circuit Breaker Sharing Network:
More than one Wall Connector is set to Master.
Set one of the Wall Connectors to Slave.

This is the code I have when I execute the TWCManager.py:
TWC Manager starting as fake Master with id 7777 and sign 77
22:48:49: Send master linkready1
Tx@22:48:49: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
22:48:49: Send master linkready1
Tx@22:48:49: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
22:48:49: Send master linkready1
Tx@22:48:49: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
22:48:49: Send master linkready1
Tx@22:48:49: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
22:48:49: Send master linkready1
Tx@22:48:49: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
22:48:49: Send master linkready2
Tx@22:48:49: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
22:48:50: Send master linkready2
Tx@22:48:50: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
22:48:50: Send master linkready2
Tx@22:48:50: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
22:48:50: Send master linkready2
Tx@22:48:50: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
22:48:50: Send master linkready2
Tx@22:48:50: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
Can anyone help me on this.
Thanks.

Scyler6411, 2019-11-19:
Not sure if  exactly identical, but the issue I had previously was similar in the sense that my TWC Master (the raspberry pi 3B+) had sent all the linkready messages but received zero response. I later found out my issue to be a faulty USB - RS-485 cable. Swapping to a new one worked like a charm.

For reference, the new cable is the 1.8m variant of this model:
https://www.ftdichip.com/Products/Cables/USBRS485.htm

I connected the orange end to D+ IN, yellow end to D- IN, and taped the other ends with electrical tape.

If you can't somehow verify that your RS-485 cable is working properly (with an oscillloscope probe to decode RS-485 signals, for example) then maybe grab a new cable instead?

llamad911, 2019-11-19:
At this stage, any idea can be a good idea to test.
I already have exactly the model USB - RS-485 that you say.  When I connect it into the usb port, a white led light on and off immediately.  Is it your case or does the led stays on ?
I don't have access to an oscilloscope, but I measured the voltage and I have 5 volts between the 5V wire and ground (red and black) and I have something around 2.3V between D+ and ground and D- and ground. My thought was "if I see voltage, the device should'n be died", but it is not enough strong to certify everithing is working well with the device.
I will also try to turn the rotary selector to another position beside the "F".
As the led message on the Power Connector sais that "More than one Wall Connector is set to Master", Maybe the selector is not precise and a position just beside can be the real "F".

Nevertheless, I already ordered another USB - RS-485 device and will see when I will receive it.

Thanks for your help.

Scyler6411, 2019-11-19:
No worries! Your voltage seems to be working perfectly fine, actually - attached is an oscilloscope output of my cable capturing linkready2 perfectly - as well as showing a max V value of ~2.24V. Spot on!

[image]

You should ONLY need the orange (D+ IN) and yellow cables (D- IN), though - the others are not necessary and should be insulated somehow so that it wouldn't short and/or interfere with other important circuitry in your wall charger.

> When I connect it into the usb port, a white led light on and off immediately. Is it your case or does the led stays on ?

My cable blinks green/red LEDs on the USB port end twice when connected. When the python script is started, red blinks signify Tx (so it blinks maybe nine times for the linkready messages), and green blinks signify Rx, so if your TWC slave and pi is set up properly, you should be able to see green blinks, which means that the pi is indeed receiving RS-485 responses from the TWC slave.

If you do not get the red LED blinks on the USB cable when you start the python script, make sure the **_rs485Adapter_** variable in your TWCManager.py code points to the right device (**_'/dev/ttyUSB0'_** usually, run **_la /dev_** on Raspbian CLI to verify the name)

TL;DR your cable seems fine afaict, but insulate other cable ends, check D+ and D- connection, check rotary selector is indeed on "F", and lastly check your code.

llamad911, 2019-12-03:
Hello,
I just received my new USB-RS485 adapter.
I plugged it and everything works perfectly.

I have a cable run of 25m (Shielded Cat5e) between my py and the TWC.
I used 1 pair for D+ and another pair for D-
Should I follow the instructions found in the TWCManager.py:
Put a 120 Ohm resistor between D+ and D-. Put a 680 Ohm (pull-up) resistor between 5v and D+. Put a 680 Ohm (pull-down) resistor between Ground and D-.
Or
Should I only use the D+ and D- cables even if I have 25m cable long?
I took care to connect one end of the drain cable to the ground of the TWC.
Thanks for your answer.