dracoventions/TWCManager #18: TWC Manager Power available for all TWCs: None, Slave available, but no power

revgazza, 2020-01-18:
I am trying to get TWCManager working. The data connection works as the slave is found. Switch is set to 'F'.
I have also set up an API to get solar data from effigy engage hub and edited the script as appropriate. 

However, this is what I am getting when I connect to my TWCManager:

Power available for all TWCs: None
TWC 9119: No power available.

I appear to get heartbeats, but nothing else happens.
Also, I have a red light slowly flashing on the TWC.

Any assistance would be appreciated.

This is what putty displays..

p_@TWCManager:~ $ python3 TWC/TWCManager.py
load_settings: nonScheduledAmpsMax set to -1
load_settings: scheduledAmpsMax set to -1
load_settings: scheduledAmpsStartHour set to 0.0
load_settings: scheduledAmpsEndHour set to 0.0
load_settings: scheduledAmpsDaysBitmap set to 127
load_settings: hourResumeTrackGreenEnergy set to 0.0
load_settings: kWhDelivered set to 119.0
load_settings: carApiBearerToken set to ****  [DATA DELETED]
load_settings: carApiRefreshToken set to ****  [DATA DELETED]
load_settings: carApiTokenExpireTime set to 1582059142.0
load_settings: homeLat set to 10000.0
load_settings: homeLon set to 10000.0
TWC Manager starting as fake Master with id 7777 and sign 77
10:08:33: Send master linkready1
Tx@10:08:33: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:33: Send master linkready1
Tx@10:08:33: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:33: Send master linkready1
Tx@10:08:33: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:33: Send master linkready1
Tx@10:08:33: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:33: Send master linkready1
Tx@10:08:33: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:33: Send master linkready2
Tx@10:08:33: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:34: Send master linkready2
Tx@10:08:34: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:34: Send master linkready2
Tx@10:08:34: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:34: Send master linkready2
Tx@10:08:34: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:34: Send master linkready2
Tx@10:08:34: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
Rx@10:08:37: () FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55
10:08:38: 32.00 amp slave TWC 9119 is ready to link.  Sign: 3D
10:08:38: Set slave TWC 9119 protocolVersion to 2, minAmpsTWCSupports to 6.
Tx@10:08:38: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:39: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:40: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:41: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:42: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:43: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:44: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:45: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Rx@10:08:45: (FC) FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55
10:08:46: 32.00 amp slave TWC 9119 is ready to link.  Sign: 3D
Tx@10:08:46: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Rx@10:08:46: (FC) FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55
10:08:46: 32.00 amp slave TWC 9119 is ready to link.  Sign: 3D
Tx@10:08:46: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Rx@10:08:47: (FC) FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55
10:08:47: 32.00 amp slave TWC 9119 is ready to link.  Sign: 3D
Tx@10:08:47: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Rx@10:08:47: (FC) FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55
10:08:47: 32.00 amp slave TWC 9119 is ready to link.  Sign: 3D
Tx@10:08:47: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Tx@10:08:48: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0

ngardiner, 2020-02-02:
Hi @revgazza,

Does this part just repeat continuously:

10:08:47: 32.00 amp slave TWC 9119 is ready to link. Sign: 3D
Tx@10:08:47: C0 FB E0 77 77 91 19 00 00 00 00 00 00 00 00 00 78 C0
Rx@10:08:47: (FC) FD E2 91 19 3D 0C 80 00 00 00 00 00 00 00 00 55

It seems that you have a comms problem of some sort, it looks like your TWCManager can see the slave TWC, but the slave TWC isn't getting your replies correctly. 

First thing I would check is the connections - is there any issue with the TX connection (from your RS435 terminal)?

revgazza, 2020-02-04:
Thanks ngardiner.
I am away at the moment, but I'll have a look at that. The RS485 might be the issue. I purchased it from Jaycar Electronics. It is actually a USB Port to RS-485/422 Converter with Automatic Detect Serial Signal Rate. I'll check it with a cro.

Garry