dracoventions/TWCManager #19: Export charging status to other program

blooby78, 2020-02-07:
Hello,

Thanks for your project it work's perfect.
I would like to know how simpli get the status off power delivery (not charging/ incharge with x amp / error ...)
You use IPC but the anwswer is direcly provide to php who is interpreted.
I would like to use bash or python program to get this status. 
Do you have a suggestion ?
Thanks
Jerome

juamiso, 2020-04-23:
I am using curl and python request lib to communicate with other systems (iobroker = home automation, and influxdb for nice graphs)

Maybe you can get some inspirations from this lines:

```
    # displaying download stats. -m 60 prevents the whole
    # operation from taking over 60 seconds.
    #greenEnergyData = run_process('curl -s -m 60 "http://192.168.13.58/history/export.csv?T=1&D=0&M=1&C$
    Photovoltaic = 0
    Smartmeter=0
    HouseBatt= 0
    url_string = 'http://<your_server>/write?db=<your_influx_db>'
    data_string = 'TWCMagager.CurrentChargingPower,from=TWCManager value=' + str(total_amps_actual_all_twcs()*225)
    myToken = '<your_influxdb_auth_token>' 
    header_string = {'Authorization': 'Bearer {}'.format(myToken)}
    ip='192.168.0.71' #This is the local ip from your inverter being connected via TCP Modbus
    client = ModbusClient(ip,port=502)
    client.connect()
    try:
      Photovoltaic = int(BinaryPayloadDecoder.fromRegisters(client.read_holding_registers(40092-1,2,unit$
      SmartMeter = int(BinaryPayloadDecoder.fromRegisters(client.read_holding_registers(40098-1,2,unit=2$
      HouseBatt = run_process('curl -s -m 5 "http://<local_iobroker_server_ip>/getPlainValue/vis.0.Batt_Discharge$
      req.post(url_string, data=data_string, headers=header_string)

```

ngardiner, 2020-05-08:
You could take a look at my fork of this, which has modules to talk to multiple external interfaces including HomeAssistant and MQTT

https://github.com/ngardiner/TWCManager