dracoventions/TWCManager #20: Tesla Gen 3 Wall Connector (WiFi)

darkmatter08, 2020-02-26:
Is there anyone interested in adding support for the Tesla Gen 3 Wall Connector (WiFi), released in Jan 2020? I see two steps:
First, the API would have to be reversed -- possibly some kind of web endpoint accessible via simple GET/POST requests, given the browser based breaker/amp selection. Or perhaps there is telnet/ssh access. Currently, there is no power sharing logic enabled, so there is no device to device traffic to watch over the network with Wireshark.
Second, the core logic for green charging from this repo should be ported over to use the new API.

Currently I don't have access to the Gen3, otherwise I'd give this a shot.

darkmatter08, 2020-03-02:
Looking at the TeslaMotorsClub forum, the author of this repo (WinterDragoness) has a Gen 3 unit and is investigating. However, there's no obvious solution so far...

> I will confirm the rs485 ports on gen3 do not output anything during normal operation or respond to anything I've tried sending. I don't know why the ports are even there though I theorize they want the option to do something with them in the future, if they ever get around to it.
> 
> The manual does not say anything about the rs485 ports but it does say there will be wifi power sharing available "in a future firmware update". No idea when.
> 
> Since you can change circuit breaker size over wifi, I tried doing that while it was charging in order to get some control over charge rate, but it seems to turn the wifi off after it's been on for 5+ mins. So wifi config can only be used for provisioning, not during operation.
> 
> Internal components of gen2 TWCs were made by Delta, but gen3 internals are all stamped Tesla. CPU is different, there is no obvious JTAG connector, and so the firmware must be a complete rewrite for the new hardware. My best guess is that Tesla is trying to build a walled garden where their products talk to each other over Wifi but they don't want anyone else intruding on the conversation. I'm not sure how difficult that might be to break into but I can tell you that I won't be able to help with the process due to agreements I have with the company I now work for. I'm also concerned that even if someone breaks into the conversation, Tesla may release firmware updates that lock people out again.
> 
> Bottom line, TWCManager will likely not get updated to support gen 3 TWCs unless someone else can figure out how to do it. Until Tesla releases a firmware update, even power sharing between two stock gen3 TWCs is not currently possible.

Copied from the [TMC forum](https://teslamotorsclub.com/tmc/posts/4502118/).

ms100austria, 2021-07-19:
Please add V3 Support. 

You can access the data via the API:

**http://<ipaddress>/api/1/vitals**

`{„contactor_closed“:false
„vehicle_connected“:false
„session_s“:0
„grid_v“:230.1
„grid_hz“:49.928
„vehicle_current_a“:0.1
„currentA_a“:0.0
„currentB_a“:0.1
„currentC_a“:0.0
„currentN_a“:0.0
„voltageA_v“:0.0
„voltageB_v“:0.0
„voltageC_v“:0.0
„relay_coil_v“:11.8
„pcba_temp_c“:19.2
„handle_temp_c“:15.3
„mcu_temp_c“:25.1
„uptime_s“:831580
„input_thermopile_uv“:-233
„prox_v“:0.0
„pilot_high_v“:11.9
„pilot_low_v“:11.9
„session_energy_wh“:22864.699
„config_status“:5
„evse_state“:1
„current_alerts“:[]}`

**http://<ipaddress>/api/1/wifi_status**

`{„wifi_ssid“:„RlJJVFoh4565464IEZvbiBXTE456IDczOTA=“
„wifi_signal_strength“:100
„wifi_rssi“:-35
„wifi_snr“:62
„wifi_connected“:true
„wifi_infra_ip“:„192.168.178.149“
„internet“:true
„wifi_mac“:"98:ED:5C:8C:77:21“}`

**http://<ipaddress>/api/1/lifetime**

`
{"contactor_cycles":32
"contactor_cycles_loaded":0
"alert_count":5
"thermal_foldbacks":0
"avg_startup_temp":5199147.0
"charge_starts":32
"energy_wh":89012
"connector_cycles":5
"uptime_s":1297280
"charging_time_s":33152}
`

**http://<ipaddress>/api/1/version**

`
{„firmware_version“:„21.8.5+g51eba2369815d7“
„part_number“:„1529455-02-D“
„serial_number“:"PGT21106012345“}
`

deece, 2021-07-20:
I'm not convince there's a big value add to support logging data without being able to control the charging.

Since you already have a nice REST interface, I'd just push that directly into OpenEnergyMonitor/HomeAssistant/whatever and call it job done, at least until we have some way to control the charging.

joriws, 2021-08-22:
At least a new firmware upgrade to 21.29.1 has power sharing api calls. So maybe twc manager could use that to control gene live.

But first api should be traced for handshakes and messages and then twc gen3 emulator could be build.

bgrigoriu, 2021-09-04:
@Darosnl Can you give more details on what is accesible on each chanel ?

Darosnl, 2021-09-07:
Could you explain what you need or what i can do?

bgrigoriu, 2021-09-07:
Hi,  
For the need , as most of us, I would like to be able to modulate the charging curent intensity depending on an external parameter ( for me is the output of my enphase solar system) . But this can be extended and used for load shedding. For the moment I could not found any doc on how to communicate with the Gen 3TWC. Only some reading is documented. Did you found anything else ?

Darosnl, 2021-09-07:
I'm happy to help but if you can point me to some tooling i can use.
Or maybe we can do it together?

bgrigoriu, 2021-09-07:
Hi,
I am happy to work with you but I really dont have the expertise to look/search what a gen3 is sending/expecting in order to modulate the charging curent. 
I think it will need that someone has two Gen3 TWC configured as "master" and "slave" and see what is transmited over the wifi connection between the two ( I do not know even if that is possible or the degree of security involved here). 
My Gen3 TWC connects as a station over the wifi to the house network and then to ionternet (TESLA server I suppose). The trafic is rather important: 56Mb uplink and "é MB downlink in two days including here a firmware upgrade.
The four requests  Vitals, Wifi Status, lifetilme and version work very well but I am not aware of any other way to interact with the TWC.
The TWC also broadcasts an Wifi AccesPoint . You can change here the charge setiings. But I saw no doc over internet about it on how to interfere with this.
If you have infos or ideas how to get it I am open

r151809, 2021-10-02:
Would the TWC access point stay up, if you connected and kept the connection alive forever? I.e. connect to TWC SSID, and send a get request for the config page once per minute or so. 
Of course the api would be better, but it's still a mystery..

mjkapkan, 2021-12-08:
Hi, Good discussion you have going on here. I recently bought gen3 as well. My initial idea was to use an openwrt wifi router with 2 bands that would forward the requests to the dedicated wifi network, that way the public part of API would be always accessible. Of course, one needs to first reverse engineer the load balancing protocol to control the charger ;).

petekelly, 2022-03-11:
This is really interesting. I have a gen3 and we’re looking to have an enphase solar system installed. Ideally I’d monitor the enphase via API calls and enable 6A charging to the car if the correct amount of excess solar is detected.

@bgrigoriu did you have any luck or success with anything? My initial train of thought is a device which is connected to the wall charger Wi-Fi hotspot and just posts charging rate updates as the solar capacity becomes available.

I guess there is also the issue of telling the car to stop/start charging too but 1 hurdle at a time!

bgrigoriu, 2022-03-13:
@petekelly
Hi,
the short answer is no, I could not do anything more. It is so far to my job today and i had no time to spar for this project. My enphase system is  running and I managed to write the routine to get all the data from the Envoy metered. It is rather simple. I did it on a ESP8266 in C++.  I had no time and experience to access the Tesla API to see if I can modulate charge intensity. The API comands for starting charging and modifyng the charging current have been published but since I am a newbee everything takes an enourmous time.

mattiasottosson, 2022-03-16:
@petekelly, if you have a tesla only, you can set the amperage via the vehicle api. I installed a twc3 in my cabin yesterday and made a really ugly python test script in home assistant in order to set the charging rate based on whats available on each phase. Tends to work fairly well, but don't know the rate limits on the api. Would be better to do it to the twc directly though since it would work with all vehicles and less things that could break along the way. 

[image]

`vehicle_connected = hass.states.get('binary_sensor.tesla_wall_connector_contactor_closed').state == 'on'
charger_state_charging = int(hass.states.get('sensor.tesla_wall_connector_state').state) != 9
vehicle_id = hass.states.get('binary_sensor.ocisly_online_sensor').attributes['id']

is_charging = vehicle_connected and charger_state_charging
max_amps = 16
charger_max_amps = 10

l1_max = max_amps - float(hass.states.get('sensor.current_phase_1').state) + float(hass.states.get('sensor.tesla_wall_connector_phase_a_current').state)
l2_max = max_amps - float(hass.states.get('sensor.current_phase_2').state) + float(hass.states.get('sensor.tesla_wall_connector_phase_b_current').state)
l3_max = max_amps - float(hass.states.get('sensor.current_phase_3').state) + float(hass.states.get('sensor.tesla_wall_connector_phase_c_current').state)

current_max_amps = int(sorted([l1_max, l2_max, l3_max])[0])
current_max_amps = charger_max_amps if current_max_amps > charger_max_amps else current_max_amps
if current_max_amps < 0: current_max_amps = 0

last_current_max_amps = int(float(hass.states.get('input_number.wall_charger_amps').state)) 

if current_max_amps != last_current_max_amps:
    hass.services.call('input_number', 'set_value', { 'entity_id': 'input_number.wall_charger_amps' , 'value': current_max_amps })

    if is_charging:
 #       logger.warning('Current max amps: ' + str(current_max_amps) + ' | Last Value: ' + str(last_current_max_amps) + ' | Is Charging: ' + str(is_charging) )
        try:
            hass.services.call('tesla_custom','api', service_data={ 'command': 'CHARGING_AMPS', 'parameters': { 'path_vars': {'vehicle_id': vehicle_id }, 'charging_amps': current_max_amps}})
        except:
            logger.warning('Could not set charge state')

bgrigoriu, 2022-03-16:
Hi,  
These is very good news. I read about the api commands but never  never used them. 
What is the minimal hardware that can log onto the tesla site and then send the commands ? (for which the suitable libraries exists/are published) .  
I implemented the read of the excess solar production on a ESP8266 ( ESP32 also works). So i would like to put all this people in a small din unit that can be integrated in the electrical distribution panel. 
Do you know if an ESP8266.ESP32 can connect or we nees a respberu pi ? à à zero or a zero 2 would work ?)

mjkapkan, 2022-03-16:
I use a cheap model with custom OpenWrt firmware.

petekelly, 2022-03-16:
> @petekelly, if you have a tesla only, you can set the amperage via the vehicle api. I installed a twc3 in my cabin yesterday and made a really ugly python test script in home assistant in order to [...]

Yes Tesla only (albeit it's not arriving until Saturday...). This was my plan actually, to 

1. Connect to the solar Enphase API to determine the current excess generation
2. If excess >6 amps (1.4KW @ 240V), instruct the car to begin charging at 6amps
3. If excess <6 amps, instruct the car to stop charging.

bgrigoriu, 2022-03-16:
@mjkapkan: that is a lot in term of size. I would like to put it on avery small hardware  so this can be included in an distribution panel. However I am not able to find a tesla V3 authorisazation library. 
@robot256 could you describe your hard/software more in detail

robot256, 2022-03-16:
I forget exactly what my setup was.  I think it was OpenHab using the Tesla API to control the car's on-off, and TWCManager on a Pi Zero controlling the Gen2 TWC amps setting, and OpenHab using HTTP calls to TWCManager.  I have a Brultech GEM power monitor feeding OpenHab with the solar and house consumption data, which is why OpenHab needed to be the center of the system.

KastB, 2022-04-13:
@petekelly It looks like you would use homeassistant.
I had pv excess charging (hopefully it is called that way in English) running for ~3 months with a python script that controls the charge speed via the tesla api and gets its information via mqtt (teslamate streaming api to reduce the api calls, vzlogger for the grid consumption from the smart-meter).
Maybe you want to have a look at here:
https://github.com/KastB/addon-tesla-pv-charging

Nevertheless I would like on "offline" version with limit imposed by the TWC better: we could have a higher rate of control and easier support of multiple cars or other brands. However, the minimal charge rate will probably be limited to 6A.

realdadfish, 2022-04-16:
@KastB Based on your work I hacked together a version without MQTT / HomeAssistant support (neither of which I have in my setup) here: https://github.com/realdadfish/addon-tesla-pv-charging - it's not final yet and not tested, just work-in-progress.

The plan is to have this running standalone on my home server. The problematic part will probably be that the vehicle is woken up each time I need to query the state of charge to determine whether or not to charge it and if so at what amperage. At night the whole process can be stopped (when plugged in) because solar_power is zero, but at day time this means the car never really gets to sleep, especially when it's not charging, and this will likely draw power.

I don't know how Teslamate gathers the current data, but I'd guess it uses the Tesla API internally as well to query it, so I don't actually know how big of a problem this becomes.

KastB, 2022-04-28:
@realdafish Probably there is a difference. There is some logic which decides when to stop polling/use the streaming api so that as little data as possible is lost while sleep mode is not prevented. Simply polling is probably not the best solution (probably ~300W higher consumption during the times sleep is prevented) https://docs.teslamate.org/docs/faq/

jonasman, 2024-05-04:
Btw gen3 also supports rs485 power sharing . Maybe this could be adapted for that bus?

KastB, 2024-05-05:
@jonasman are you sure there is full rs485 support for v3? I only read about V2 - do you have a link to detailed information?

My solution to control the charge current on the vehicle side broke due to restrictions on the API for Tesla cars.

Does anybody have any experience with this API and it's capabilities/limitations?
https://github.com/jeremyvisser/wc3magic
Cheers

jonasman, 2024-05-15:
@KastB  that’s how neurio is connected to the wc3:

https://service.tesla.com/docs/Public/Energy/Charging/Wall-Connector-Device-Setup-in-Tesla-Pros-EN/GUID-8CD32F0F-C5BA-43D9-951A-EB3E17043427.html

deece, 2024-05-15:
Nice, if someone can track down the meter model, that should lead to the protocol specs (probably Modbus), and then the meter can be faked.

frizzby, 2024-08-09:
Has there been any further progress? I do have a neurio meter talking to WC3. However I'm trying to solve the opposite problem: I want to intercept the CT readings data. I've got a rs485 and I'm going to capture some data for analysis.

KastB, 2024-08-09:
Hi,
to my knowledge, there was no further progress by now. But a capture file would be a big step.
If you could record one, it would be really great.
Any additional information would ease decoding it:
1) What is your electrical setup e.g. How many phases, which voltage do we expect, which currents are set. Probably the serial of the meter is communicated as well (maybe you want to share privatly)
2) What were the readings during your recording e.g. min/max currents
3) Timing of induced events e.g. maybe you could turn on/turn off load after x seconds, one on each phase, so that we see a step response in the data and can more easily correlate the data with the values we see.

Cheers

frizzby, 2024-08-13:
Hi,
I have captured some data with a python script. I'm using pyserial and reading bytes into a buffer one by one until there's a delay greater than 0.8s. In this case I print the collected data in binary format from the buffer and reset the buffer.
I'm in North America (split phase) and I have two neurio CTs connected to two live wires (L1, L2). Each wire to N is 120V and there's 240V between the live wires.

As you suggested, I ran several tests: 
1.  L1 load on, L2 off
2. L1 load off, L1 on
3. L1 on, L2 on
4. Disconnected neurio from power and after a short delay reconnected.

I'll attach the results below.

A few observations, in no particular order:
1. I can connect to my WC3 with Tesla One app and see meter readings (attached). 
3. Neurio meter has 4 ports for CT sensors and I see 4 readings in the app. Since I use only 2 CTs the two unused are floating and show small values that fluctuate around zero (~±0.070W). This means at least 4 values should be transmitted from the meter to WC3.
1. It doesn't appear that neurio transmits its serial number (SN) after I power cycle it. From the captured data I don't see anything that looks like the SN. Also, I have a spare neurio meter and when I swap one for another Tesla One app still shows the old SN.
4. Unless I'm missing something obvious I'm pretty sure it's not just modbus. I've tried interpreting the captured data as modbus protocol (RTU mode as described [here](https://www.modbus.org/docs/PI_MBUS_300.pdf). I didn't get good results. The data seems to be changing too randomly. A change in load doesn't correlate with the changes I see in the captured data (small change in load -- big change in the data).
5. A message is transmitted every ~1s.
6. Precision. The app shows power with up to 3 digits after the decimal point (e.g. 1254.984W).
7. Values in the app are updated every ~2s.
8. If I disconnect the WC3 end of the rs485 line and keep the neurio meter connected all communication stops. Which means there's a two-way communication. Or the meter somehow else knows it was disconnected.
9. The value for power is signed. If I flip a CT the power is displayed as a negative number.  
10. The app also shows current (in A), however it appears to be always displayed as a whole number and rounded up (e.g. 272W (3.0A)). I'm not sure if this value is transmitted or is just calculated by the app.
5. I looked at the Tesla One iOs app. It's made using React Native and most of the code is in plain text. I was able to capture wi-fi packets between my phone and the WC3 while looking at the neuri meter readings in Tesla One app. They use protocol buffers for communication between the app and WC3. I'll attach an example of the decoded protobuf below.
6. I have a hunch IEEE 754 could be used for float encoding. 
7. Despite the fact that the data doesn't look as modbus to me, everything else suggests it should be modbus: tesla documentation, manuals, also Tesla One app has multiple mentions of modbus in the context of Neurio and WC3.

The python script I'm using https://gist.github.com/frizzby/5582fc761e614c0c494c11f71e4e19eb

[image]
[image]

```
    4 { // WCMeterInterface                                                                         
      1 { // NeurioMeterInterface meterInterface                                                                       
        1 { // NeurioMeterConfig config                                                                     
          1 {                                                                    
            6: 0x36303532                                                        
          }                                                                      
          2: "VAH4770AB5642"                                                     
          3 {                                                                    
            1: 7 // int32 location                                                                 
            2: 0x3f800000 // float realPowerScaleFactor                                                      
          }                                                                      
          3 {                                                                    
            1: 7 // int32 location                                                                
            2: 0x3f800000 // float realPowerScaleFactor                                                       
          }                                                                      
          3: ""                                                                  
          3: ""                                                                  
        }                                                                        
        2 { // NeurioMeterConnection connection                                                                       
          1: 3 // connectionStatus                                                                  
          2: 1 // connectionError                                                                   
          3 { // Rssi                                                                   
            1: 200                                                               
          }                                                                      
          4: "1.6.1-Tesla"  // firmwareVersion                                                      
[... cut ...]
0010100000111000001000101011111011111100
001010000111001111111011011010000011000011100100
001010001011101011111111001010000111110011111110
001010000110000111111111011010000011000111100100
001000000001001111111111001000000011101011111011
001010000010110011111111001000101100010011111001
011000101010011111111110001000101011111110011000
00101000111000100011000000010010
0011000011101100111111110010000010011100
001010000110000111111101001010001010110111111101
001100000110001111111111001000001001000111111111
```

frizzby, 2024-08-13:
I would appreciate it if someone else could take a look at the data. 
Clearly, the load values are being transmitted. There must be at least 4 floats or ints (if milliwatts are used). And I don't think the data is encrypted – it wouldn't make sense.

r151809, 2024-08-13:
Exellent job @frizzby :)

Serial number and such are probably read by WC3 when provisioned. So we should record the data from the installaion phase. I assume that the WC3 will not just blindly ask the data - it first reads the model and serial to verify it really is Neurio.

Not really done decoding yet, but the first hunch is that it could be RTU Framing. Three ones at the begining looks like 3.5 character start, the rest would be missing from capture.. Some documentation here: 
https://www.modbustools.com/modbus.html

KastB, 2024-08-13:
I didn't have time to dig into it today.
The sequences without meter are interesting. I would guess those are requests to the meter and we see the response data from the meter in the "normal" data. Thus we might be able factorize and only deal with those smaller chunks between those requests.

KastB, 2024-08-14:
@frizzby Great work, thank you! I had a second look into it and wondered if we really can expect so little data:
There is about 48bit, sometimes only 40bit of data. If I'm not wrong, the 7 digits of power from your screenshot need 32bits twice already. I'm wondering if the baudrate was higher / changed during your recording.
Did you try other baudrates / how did you determine it?

frizzby, 2024-08-15:
Thanks everyone for taking time to look at it!

@KastB, aaah, yes, I think you're right – I'm not using the right baud rate.

For some reason I assumed (incorrectly) that since I'm getting any data it means the baud rate is right. Now that I think about it that assumption makes no sense.

I don't have an oscilloscope, but I do have a logic analyzer. Can it be used to determine the baud rate? In my understanding it should be possible if used with rs485-TTL converter, not directly from rs485 line. Can someone confirm?

Meanwhile, I'm going to just try different baudrates to see what results it will give.

frizzby, 2024-08-15:
Here's how it looks with 115200 https://docs.google.com/spreadsheets/d/1KzO4i0zNmd9PlaaSO53OKI9kKUAVvXDLb3r9rOAyFCA/edit?usp=sharing

frizzby, 2024-08-15:
Disconnected and reconnected the meter. 115200 baud rate.
```
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111
0000000100000011000000001000100000000000000010100100010111100111000000010000001100010100010001000100001010001110011001100100010010001000100111001101111010111101000000010110111100100101101111100010001000110100101011111100001101110010111000110100100000001101111100000000000100000011000000001111010000000000000010000000010111111110000000010000001100010000010000001101110100000101000010100100000100011011110111111110110000111100110000110101001111111000001111001100101110100011000001100010100000011100
0000000100000011000000001000100000000000000010100100010111100111000000010000001100010100010001000100001100010001000000000100010010001010001010110110010110111101000000010110111100100101101111010000000111000011101111111100001101110011000100111110000100100111011010000000000100000011000000001111010000000000000010000000010111111110000000010000001100010000010000001101101110111000100011010100000100011100100001001011001100111100101110110101011100110010001111001110001011110011100001011111111111111100
0000000100000011000000001000100000000000000010100100010111100111000000010000001100010100010001000100010010101001000001000100010010001000110110100010010010111101000000010110111100100101001111010000000111000011101111111100001101110011000000111010111101111111101111010000000100000011000000001111010000000000000010000000010111111110000000010000001100010000010000001101111011011011011001000100000100011100100111000111000000111100110011100011100001001011001111001100101110100011000001101110001010000101
0000000100000011000000001000100000000000000010100100010111100111000000010000001100010100010001000100001001100001110110100100010010001000101100110010011010111101000000010110111100100101101111011000000111000011101111111100001101110010110100110001010100011010110000010000000100000011000000001111010000000000000010000000010111111110000000010000001100010000010000001101110001000000110010100100000100011011101010001111111000111100110001111010111101001101001111001011110111001011001111011001100000100110
```

frizzby, 2024-08-15:
This is how it looks with 115200 with low vs high load https://docs.google.com/spreadsheets/d/1KzO4i0zNmd9PlaaSO53OKI9kKUAVvXDLb3r9rOAyFCA/edit?usp=sharing

frankenbubble, 2024-08-15:
looks like modbus RTU, but all the responses have an extra 107 bytes after the frame

convert to hex and use a tool  like https://npulse.net/en/online-modbus  (enter hex into Hexadecimal Translate, hit enter etc)

request as hex 01030088000A45E7
a response as hex (first 15 bytes, removing the 107 extra bytes)  010314444261DA4488B326BD016F25BD81C3BFC372D3151AC1

CRCs seem to check out , not sure what the extra 107 bytes are, not timestamps that I can see

First few bytes of each extra data portion:
Response 1: 01006f25be2234afc372
Response 2: 01006f25be4038efc373
Response 3: 01006f254f4038efc373
Response 4: 01006f25be6038efc372

frankenbubble, 2024-08-15:
here is some python that reads your binary and decodes it.  seems they are 32bit floats, and match your readings ?

https://gist.github.com/frankenbubble/33d14a01c65c80f42e0f92a09127a888

using the raw data in your sheet, putting it data.txt for the script I get this output

Line 1:
Request:
Raw Hex: 01030088000a45e7
Address: 1
Function Code: 03
Data: 0088000a

CRC (received): E745
CRC (calculated): E745
CRC Match: True

Response:
Raw Hex: 01031442ab3a1643c200c9bd016f253d81c3bfc233b4bcef3a
Address: 1
Function Code: 03
Number of bytes: 20
Register Data: 42ab3a1643c200c9bd016f253d81c3bfc233b4bc
Interpreted Data:
16-bit Registers: [17067, 14870, 17346, 201, 48385, 28453, 15745, 50111, 49715, 46268]
32-bit Floats: [85.61344909667969, 388.0061340332031, -0.03160013630986214, 0.06336163729429245, -44.92649841308594]
CRC (received): 3AEF
CRC (calculated): 3AEF
CRC Match: True

----------------------------------------

your sheet indicates its for reading 86W and 380W, which looks like a close match to the first 2 32-bit floats in the response

for the page indicating 2000KW , script says 

32-bit Floats: [1985.830322265625, 2206.439697265625, -0.03160013630986214, -0.1267232745885849, -52.32987594604492]

KastB, 2024-08-16:
Nice script.
I just made a small extension to parse the second request/response for each timeslot
So now we have the power as well as the current readings:
https://github.com/KastB/twc_meter_communication/blob/master/modbus_test.py

KastB, 2024-08-16:
This script give as csv to show the Power/Current values:
https://github.com/KastB/twc_meter_communication/blob/master/modbus_read_data.py

KastB, 2024-08-16:
@frizzby I guess your goal should already be achieved. For us/me there are still 2 or 3 open points:
1) What are the last 32-bits in the power message for (maybe frequency/crc/voltage?)
2) How does the initial commissioning look like so that we can emulate a meter. (Could you maybe do a recording of that as well?)
3) Does the TWC completely shut off, if the reported currents limits are reached?

KastB, 2024-08-16:
1) For information ab to each clamp, there probably is too little information. Additionally: when we calculate voltage from power and current, for the low power recording, the voltage seems to be too low. So either there is pretty bad fluctuation on the grid, or a scaling factor involved. Maybe a longer recording could help there.
2) Thats my motivation as well. 
3) This would be bad news for me. I would like to do PV charging, and thus need the ability to completely stop charging by setting a high current, so that the limit seems reached for the charger.

@frizzby It would be great if you could 3 more tests:
1) A long recording, where different load scenarios happen, without interrupting the recording (I think we can find the transitions now), so that we maybe can decode the 5th power value, which might be a scaling factor
2) Do a recording of the initial setup, so that we could create a virtual meter
3) It would be great if you could verify the behavior of  the charger, when the limit of the cabinet is reached => maybe configure a small current limit for the cabinet, create load so that the limit is reached, and check if charging stops, or if it is only reduced to 6A

frizzby, 2024-08-17:
Thanks everyone for the help! Nicely done decoding it!

1. I'm not sure what are the last 32-bits in the power message. it usually sits at around -50. When I start EV charging it drops to -240 and doesn't react much when I increase the load even more (by turning on the oven). However, it jumps to 1200 when I start my AC. What could it be? Reactive power? See https://snapshots.raintank.io/dashboard/snapshot/WbI32Gl5jiOXmE464csvNzdyRwYjCfqQ
3. There's no way in the app to unlink/remove the meter. I'm going to Factory Reset the wall connector and do recommissioning while recording the data from rs485 line. I will post the results separately.
4. The lowest limit I can set is 40A. The app doesn't allow lower values. Unfortunately the charging doesn't stop when limit is exceeded. It is only reduced to 6A. The same happens when I disconnect the meter.

Btw, my wall connector is Tesla Universal Wall Connector, not TWC3. But I think they are the same software-wise.

KastB, 2024-08-17:
1) This could really speak for reactive power: Unaffected by the resistance load of the oven (if you don't have an induction stove), capacitors of the EV charging would be in-line with more negative reactive load, and the induction of the motors in the AC could be correct positive reactive power. From what we learned in theory, this could very well be reactive power and it would make sense to place it close to the loads for each clamp. I never had a look on real-world reactive power data, so I wouldn't know for sure, but it sounds plausible.
2) That would be great!
3) That really is a pity. Maybe there is a 2nd limit when we greatly exceed the limit, which we could reach with the virtual meter. Otherwise we need manual intervention to stop charging at least at night.

Thank you very much for your efforts doing those recordings!

johado, 2024-08-17:
Is it clearing whats needed in the network side towards the WC3 to enable and configure the load balancing without an actual Neurio meter?
(Protobuf messages mentioned)
Does the vitals or other REST endpoint on the WC3 show anything related to load balancing or is it all protobuf?
(I currently do my load balancing towards my ca using the Tesla vehicle-command go program from an RPi using BLE so I don’t use the cloud API:s, but controlling the charger would be better.

frizzby, 2024-08-18:
Unfortunately I didn't have time today to do the recommission, and tomorrow I leave for vacation for a few weeks. I have some vested interest myself in figuring out the rest of the modbus communication, so I will recommission and record the comms when I'm back home.

frizzby, 2024-08-18:
> Is it clear whats needed on the network side towards the WC3 to enable and configure the load balancing without an actual Neurio meter? (Protobuf messages mentioned) Does the vitals or other REST endpoint on the WC3 show anything related to load balancing or is it all protobuf? (I currently do my load balancing towards my car using the Tesla vehicle-command go program from an RPi using BLE so I don’t use the cloud API:s, but controlling the charger would be better.

Protobuf mentioned above is an excerpt from the communication between TWC and Tesla One app with the phone connected to the AP the TWC is broadcasting. Once we're able to emulate the meter TWC should automatically detect the meter and you would use Tesla One app to configure CTs as Conductor and set the current limit. You can see how that looks in the [video](https://youtu.be/ZLZFYgo6OZk) and in the [pdf](https://digitalassets.tesla.com/tesla-contents/image/upload/NA_Wall_Connector_Smart_Current_Limits_App_Note_EN.pdf).

I didn't see any changes it `/api/1/vitals` endpoint after configuring the meter.

KastB, 2024-08-18:
@johado I only read about the tedapi which seems to exist for several products: [PowerWall](https://github.com/jasonacox/Powerwall-Dashboard/discussions/392) [Or TWC3](https://github.com/jeremyvisser/wc3magic)
But I didn't test it by now.

KastB, 2024-08-18:
[Here](https://github.com/jasonacox/Powerwall-Dashboard/discussions/392#discussioncomment-8445232) they have a reactive power for each clamp

frankenbubble, 2024-08-18:
afaik, the wall chargers don't expose any data like this via the 'normal' internet wifi, although you can get some different data via the built in 'setup' wifi, but that still goes asleep after some minutes.
the powerwalls seem to have always on wifi that gives extend information, which i think is what they are looking at.

I suspect the 5th float is indeed (total) reactive power, as it make sense with the AC turning on it becomes a high positive value, otherwise a small negative value.    I also suspect if we queried the meter at other addresses we would get more data, and we might see that happening in the setup communications.

it would be interesting to see what serial requests come from the TWC3 during setup as it looks for a meter (what address it queries and with what codes etc),

frankenbubble, 2024-08-21:
I hooked up my twc for serial, and notice it regularly outputs a modbus request, adding here as information, i assume a probe for a meter.

Binary: 00000001 00000011 10011100 01000010 00000000 00000110 01001011 10001100
Hex: 01 03 9c 42 00 06 4b 8c
ASCII: �BK�
Received CRC: 0x8C4B
Calculated CRC: 0x8C4B
Message Length: 8
Address: 1
Function Code: 3
16-bit Registers: [40002, 6]
32-bit Floats: [-6.418924582637371e-22]

johado, 2024-08-23:
I can confirm that I get exactly the same data. So getting a trace from power on from someone with a Neurio connected would help us get one step further hopefully.

frankenbubble, 2024-09-20:
Need @frizzby or someone with the hardware to post more serial data

frizzby, 2024-09-30:
Attached are two files:
I started the recording -> clicked factory reset on the TWC -> the TWC is now in waiting for commissioning state -> I stopped the recording:  [factory_reset.txt](https://github.com/user-attachments/files/17183452/factory_reset.txt)

I started the recording -> I commissioned the TWC and configured the meter -> I stopped the recording:  [after_factory_reset.txt](https://github.com/user-attachments/files/17183460/after_factory_reset.txt)

KastB, 2024-09-30:
For the provisioning/resetting: Those seem to be ASCII Strings:
Registers: 9c420006 ASCII: ...BGenerac.  Raw: 0001004247656E6572616300 

To register this seems to be the register:
Registers: 00010037 ASCII: 0x000004714B056861..1.6.1-Tesla.........012.00020A.H....90954.VAH4810AB0231.................04:71:4B:05:68:61.

Raw:3078303030303034373134423035363836310000312E362E312D5465736C6100FFFFFFFFFFFFFFFF3031322E3030303230412E480000FFFF3930393534005641483438313041423032333100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF30343A37313A34423A30353A36383A363100

Maybe/Probably 9c420006 is just a magic string.

00010037 holds different Blocks.
Notice how "." encodes different things in the ASCII representation: 
 - "2E" when there is a "." within a string
 - "00" or "FF" to fill a block
I'm not sure where the limits of those blocks exactly are and what they represent. The first one seems to be 20 chars long and a hex number. Second one the firmware version, then maybe a configuration, then there is the serial number of the meter and at the end a mac address.
What I miss is the number under the meter (52506). For the parts besides MAC, Serial Number, and Firmware version I have no idea what they represent.

frankenbubble, 2024-09-30:
The two files seem identical, but i think it's ok as it's probably the commissioning one.

I put a simple html page together that makes parsing the binary easier.   https://frankenbubble.github.io/modbus_decoder/index.html ,  

it looks like ~20 gets of data at 2 addresses, is probably the metrics. and 2 at 2 other addresses is the setup (I'm guessing).    My TWC sends out the requests to address 0x9C42, and it looks like the response is the manufacturers name.   

the requests to 0x0001 seem to have the firmware and other information.  some of it looks different to your screenshots earlier such as VAH4810AB0231  looks like it previously in you data was VAH4770AB5642 .

frankenbubble, 2024-12-05:
I finally got to the place where my twc3 is, but had limited time to experiment.  I got as far as the twc showing the meter, when i sent a response, but with a red exclamation mark saying it had an error.  I won't be back there for a while though to look further, it could have been a electrical bus problem or code.  

I put together a script to give responses to the twc requests,  and it should be giving the same responses as per the @frizzby binary information.  code is here https://github.com/frankenbubble/twc3-modbus  , however i only see the 40002 register request over and over, so i'm guessing it didn't like me response to that.

frankenbubble, 2025-01-16:
small update,  trying again, I decided to swap out my rs485 converter .  The 'good' news is that now the TWC is responding differently. I see the initial 40002 request,  my reply, then the Tesla One app shows a meter as before, with an error.  but now I see the requests for register 1 and 136, recurring. and also 244 and the replies.

I put the output here , https://gist.github.com/frankenbubble/3d54dfcf3ca38f4ee7fda2f24326bc35

I will need to revisit, but will be weeks, just to be sure all the rs485 paths are good (am using cat5e and 12 meters run. with no resistors) 

@frizzby are you in a position to test ?

[image]

frizzby, 2025-04-25:
@frankenbubble sort of, let me know how I can help

[55 more replies]