dracoventions/TWCManager #27: Tesla Authentication Changed (January 2021) - Does TWCManager Support?

ChutneyMary, 2021-02-08:
Hi. I'm about to embark on setting up TWCManager. I have an unrelated Node-Red flow that interogates the Tesla API. I see that Telsa has recently (late January 2021), changed the authentication routine that broke a couple of Node-Red modules.

https://tesla-api.timdorr.com/api-basics/authentication

Will TWCManager be updated to manage the new authentication routine? I assume with a broken authenication routine, the ability to start and stop charging will be lost in TWCManager? Will charge rate control still be available?

deece, 2021-02-08:
I just got bitten by this and had the car draw 16kW during what would have been peak time, if I was on a time-of-use plan.

These are the relevant commits in TeslaJS that fixed it for NodeRed:
https://github.com/mseminatore/TeslaJS/commit/e95e896e21132d8921b8d2e435099227d1562310
https://github.com/mseminatore/TeslaJS/commit/e8ff2d51f63bd73f7892487cdc20c93fe79f5fe9
https://github.com/mseminatore/TeslaJS/commit/027795fe4040309330e052094e57183e0e5ed568

For now, I'm working around it in Node-Red, which already controls a few other things for me, like disabling sentry mode if the car is in the garage, and the maximum charge level (less for daily charge, more if I'm not home)

ChutneyMary, 2021-02-08:
There is also this fork of TWCManager that has incorporated the new authentication (excluding Multi-Factor Auth) -

https://github.com/ngardiner/TWCManager

https://github.com/ngardiner/TWCManager/issues/157