dracoventions/TWCManager #31: install sysv_ipc SyntaxError: invalid syntax

lkyan, 2021-11-01:
SyntaxError: invalid syntax

pi@raspberrypi:~ $ sudo python3 -m pip install sysv_ipc
Collecting sysv_ipc
  Using cached https://files.pythonhosted.org/packages/0c/d7/5d2f861155e9749f981e6c58f2a482d3ab458bf8c35ae24d4b4d5899ebf9/sysv_ipc-1.1.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-udoiaogl/sysv-ipc/setup.py", line 11, in <module>
        import prober
      File "/tmp/pip-build-udoiaogl/sysv-ipc/prober.py", line 137
        d["SYSV_IPC_VERSION"] = f'"{version}"'
                                             ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-udoiaogl/sysv-ipc/

lkyan, 2021-11-01:
debug
sudo pip3 install sysv_ipc==1.0.1

Jille, 2022-02-27:
Which version of Python are you using? fstrings are a Python 3.6 feature.