dracoventions/TWCManager #33: phases not represented / calculated correctly

Dustovsky, 2022-03-01:
something I observed today (1.2.4)

in the config.json, I edited the number of phases to "2" (because obviously, I have connected only two phases...)
Nevertheless, on the homescreen top righthand "charging*, I see only the power of one phase, although lateron the voltage of both phases is stated...

What did I wrong?

simotronic, 2022-03-01:
Well, technically you are only connecting 1 phase. The neutral isn't
connected. So the AC voltage is between the two phases or a Neutral and a
phase.
Initially, the reaction is to see L1 and L2 and consider them two phases,
which they are, except you aren't connecting a 3rd wire, the neutral.
Therefore it's only one phase..
If you had a three phase supply, then the phasing between the three phases
is part of the equation because the phases are 120 degrees out of phase
with each other. Probably easier to look at an electrical diagram which
explains the phasing!
Also, those two phases from the transformer may well be in phase, or 180
degrees with respect to each other since the transformer on the pole or
ground outside is only using one of the medium or high voltage phases from
the utility.
Be careful with what you are looking at.
Simon

-- 
This email, and any attachment to it, may contain information that is
proprietary, privileged or confidential or that may be otherwise legally
exempt from disclosure and is intended only for the individual(s) or entity
to which it is addressed.   If you are not the named recipient, or the
employee or agent responsible for delivering it to the intended recipient,
you are not authorized to read, print, retain, copy, disclose or distribute
this email or any part of it.
If you have received this email in error, please return it immediately to
the sender, delete it and all copies from your system, and destroy any hard
copies of this communication.

Dustovsky, 2022-03-01:
@simotronic 

probably I was not clear enough (english is not my motherlanguage...)

In switzerland, we have regular 3 phase supply. in my case, I'm using 3x16A for the cars, 1x16A (max 3.6kw) for the Hyundai eKona and 2x16A for the Tesla (2x 3.6kW)

Everything works as designed, actually since years.
Only the report/numbers on the TWCManager homepage are wrong and indicating only the power from one phase
example: 
I'm charging @8A
The car get's 2x8A = 3.7kW
TWCManger prints only 1x8A = 1.85kW

closed 2022-03-03 by Dustovsky

simotronic, 2022-03-04:
I see your dilemma! If the code is corrected, it should match an actual
measured power.
Simon